ms-iot/lightning #35: Why RPi2 hardware PWM is not utilized?

andriysavin, 2016-05-09:
There are lots of information on the Internet that RPi2 has hardware PWM support, and there are examples of how to use it with Linux libraries implementations (like WiringPi). On Windows 10 IoT, however, the only options are the software emulation on a GPIO pin and an external chip (and the only one is supported). So I'm wondering what prevents from implementing support for the hardware already on board?

jessekaplan, 2016-05-12:
The hardware PWM on the RPi2/3 is designed for audio purposes rather than for device control and so, while some other OS platforms do provide access to the hardware PWM, we've found that the vast majority of folk who need to use PWM on an RPi end up using software PWM or and external board. The hardware is usable for controlling basic RGB LEDs, but for things like servo motor control you almost always want something with a higher resolution than the Pi's hardware pin (and you very often need more than 1 anyways). 

With that background we have instead focused on the soft PWM model and supporting expansion boards. We added support for one of the most popular PWM expansion chips, but it should be easy to adapt to another if you have other preferences (it's all open source, and doesn't require any OS updates). 

That said, we're always re-evaluating features based on customer need and so this is something we may revisit in the future. What particular hardware were you thinking about driving with the hardware PWM pin?

Thanks,
Jesse

andriysavin, 2016-05-12:
@jessekaplan Thanks, great explanation! However, I read (e.g. [here](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=105044) ) that RPi**2** has 2 PWM channels, and only one of them is reserved for audio, so the second one should still be usable.
Maybe its true that for "serious" prototyping most use external chips, but for less complex projects it might be desirable to not connect another chip with all its circuits just to drive single motor. Regarding software implementation: I guess it can be unstable sometimes due to GCs and OS scheduling, and also can be power and CPU consuming, which is critical for battery-powered projects.

Well, these all are my general thoughts. In my concrete case all is simple: I was going to try to drive a motor with PWM, but was surprised when calling the API to get a PwmController gave me null. Then it took me several hours to understand why my Pi with some pins marked PWM can't do PWM, try get lighting provider instead, fail with using the **default** PwmControllerProvider it returned, and then look through the source code to understand that it didn't implement HW PWM either, but instead represented an expander chip. And that only **non-default** controller provider gave me the software implementation which worked. As you can see, even though UWP API is very convenient and readable, in practice all these tricky details can scare and disappoint a person why tries to do something for the first time.

jessekaplan, 2016-05-12:
Regarding the PWM channels, yes one is used for audio and the other is available from the hardware pin, but it is driven by silicon that is designed to run audio and won't give the fine grain control you need to control servo's. 

I'd highly recommend trying out the software PWM implementation as for the majority of cases it meets prototyping, and even production, needs and is what's commonly used to drive these solutions on Raspian as well. It's written in native C++ and spins up a high priority thread and so, while it does burn a core, it tends to be very stable for controlling servo motors and similar devices. GC's will not pre-empt it and, as long as there aren't too many high priority threads in your apps, it very rarely gets pre-empted and then rarely for longer than 15.8 ms. The single thread/core is used regardless of how many PWM pins or duty cycles are being used. 

I really appreciate your feedback on the PWM provider developer experience. There's not much we can do to improve the case where you have no providers, but I think it's definitely worth considering changing the default for the lightning provider to software pwm rather than the one that required an expander board. The trade-off would be that some wouldn't realize that they are doing software pwm rather than hardware pwm (but this could also be a feature for the broad set of people for which it wouldn't matter). What do you think? Would changing the default have helped you out here?

Thanks,
Jesse

andriysavin, 2016-05-12:
Thanks again :) I think, changing the default would definitely save me some time. It would be nice if some external devices detection was possible to make the software implementation a fallback for the case when no HW devices found. Or at least detect integrated PWM (don't know if this is possible), and then fallback to SW. The latter would be more or less sensible, because more people (I think) will understand that an external chip can't be autodetected and thus can't be the default, and will try to look at all controllers. Honestly, I wouldn't even know you created this cool abstraction for extenal chips if I didn't browse this repository. If I had an external PWM chip, I would first try to communicate with it manually via the SPI (or other) bus :)

andriysavin, 2016-05-12:
Some more thoughts, since you've asked :)
Regarding all this lightning staff. First of all, it looks unusual in that sense that this is kind of part of a driver, but is implemented in user mode (as far as I understand, the kernel part only implements device registers memory mapping into a user process). Its more common that an OS abstracts at system level api how exactly a user process communicates with a device, and a driver implements particular approach. In case of Windows IoT, not only driver switch is required to change the implementation, but also provider switch is needed at user-mode level. I don't say this is bad or significantly confusing, but makes a developer asking more questions, like why don't I need to work with providers when dealing with USB or Bluetooth, but need to bother with them when dealing with IoT buses.

Second, what concerns me more, is that lightning starts containing things which seem to be not lightning-specific. I mean concrete implementations of higher level api like PWM or ADC for external chips. Currently for PWM there are just two providers (SW and for an external chip), but imagine that in couple of years you will be supporting 10 chips. Or more. And all this code will be in the same single package, nicely hidden behind GetControllers API. Further, imagine this method returns 10 controllers, while only one is really connected. Wouldn't this be looking somewhat confusing? Enumerating devices is good when you have PnP, so you know what devices are really connected.

So my thought here is that the lightning package should contain only common API for working with buses, and I would expect each external device to be supported by a separate package (much like this is usually done for Arduino). This means that a developer explicitly adds support for a particular component, instead of choosing the correct one by a magical index in the array of controllers.

Sorry for my style of writing, its past midnight where I am :)

closed 2016-05-14 by andriysavin