ms-iot/lightning #43: I2C Device disabled when using DMMD, enabled with Inbox Driver.

dulfe, 2016-08-24:
I just started looking into the **Lightning provider** and the first thin I tried by **I2C** and I found an issue. The I2C device (returned by `DeviceInformation`) shows it as **DISABLED** but when using the *Inbox Driver** shows as **ENABLED**.

I have a RBPi2 B with Windows 10 IoT 10.0.14393.67

I enabled DMMD Driver and created a simple program that makes reference to:

- Microsoft.IoT.Lightning: 1.1.0
- Microsoft.IoT.SDKFromArduino: 1.1.1

Also enabled the extensions:

- Windows Desktop Extensions for the UWP: 10.0.14393.0
- Windows IoT Extensions for the UWP: 10.0.14393.0

Modified Package.aspxmanifest with 

```xml
<Package
  xmlns="http://schemas.microsoft.com/appx/manifest/foundation/windows10"
  xmlns:mp="http://schemas.microsoft.com/appx/2014/phone/manifest"
  xmlns:uap="http://schemas.microsoft.com/appx/manifest/uap/windows10"
  xmlns:iot="http://schemas.microsoft.com/appx/manifest/iot/windows10"
  IgnorableNamespaces="uap mp iot">
....
  <Capabilities>
    <Capability Name="internetClient" />
    <iot:Capability Name="lowLevelDevices" />
    <DeviceCapability Name="109b86ad-f53d-4b76-aa5f-821e2ddf2141"/>
  </Capabilities>
</Package>
```

And with the following code:
```csharp
public MainPage()
{
	this.InitializeComponent();

	if (LightningProvider.IsLightningEnabled)
	{
		LowLevelDevicesController.DefaultProvider = LightningProvider.GetAggregateProvider();
	}

	CreateAsync();
}

private async Task CreateAsync()
{
	var aqs = I2cDevice.GetDeviceSelector("I2C1");

	var i2cDevices = await DeviceInformation.FindAllAsync(aqs);
	var i2cDevices_ManualAQS = await DeviceInformation.FindAllAsync("System.Devices.InterfaceClassGuid:=\"{a11ee3c6-8421-4202-a3e7-b91ff90188e4}\" AND System.DeviceInterface.Spb.ControllerFriendlyName:=\"I2C1\"");
	var allDevices = await DeviceInformation.FindAllAsync();

	foreach (var device in allDevices)
	{
		System.Diagnostics.Debug.WriteLine($"{device.Id}  - {device.Name}");
	}
}

```

`i2cDevices` does not return any devices, but `i2cDevices_ManualAQS` returns 1 which is DISABLED (I manually changed the AQS to include enabled and disabled devices) and the `allDevices` shows there is a I2C1 device in it (`\\?\ACPI#MSFT8000#1#{a11ee3c6-8421-4202-a3e7-b91ff90188e4}\I2C1`)

If I changed the driver to **Inbox Driver** `i2cDevices` has 1 device and it is ENABLED.

So, either `DeviceInformation` is returning incorrect data or the I2C device is not enable with using Lightning.

Thank you!

msalehmsft, 2017-01-24:
DMAP and the inbox drivers are mutually exclusive. The default (inbox) I2C driver as well as the SPI and GPIO ones are disabled when DMAP driver is enabled through the device portal.

So, the code above will not work. Instead please use the sample at https://github.com/ms-iot/BusProviders/blob/develop/Microsoft.IoT.Lightning.Providers/WeatherStation/StartupTask.cs :
``` csharp
            // Set the Lightning Provider as the default if Lightning driver is enabled on the target device
            // Otherwise, the inbox provider will continue to be the default
            if (LightningProvider.IsLightningEnabled)
            {
                // Set Lightning as the default provider
                LowLevelDevicesController.DefaultProvider = LightningProvider.GetAggregateProvider();
            }
            // The code below should work the same with any provider, including Lightning and the default one.
            I2cController controller = await I2cController.GetDefaultAsync();
```

IoTGirl, 2017-01-24:
Hello Dulfe,
I am closing this issue based on the answer given.  Please reopen if it does not address your problem.
Sincerely,
IoTGirl

closed 2017-01-24 by IoTGirl