ms-iot/lightning #47: The ValueChanged-event of a GPIO Pin with MinnowBoard MAX not working

SaschaIoT, 2017-07-28:
Hello together,

the ValueChanged event of a GPIO Pin gets never called using the lightning providers with MinnowBoard MAX.

Is the ValueChanged event still not supported with MinnowBoard MAX and Windows IoT Core 15063.0?

LowLevelDevicesController.DefaultProvider = LightningProvider.GetAggregateProvider();
var gpioController = GpioController.GetDefault();
var pin = gpioController.OpenPin(0);
pin.SetDriveMode(GpioPinDriveMode.Input);
pin.ValueChanged += (gpioPin, args) =>
{
//Gets never called
};

SaschaIoT, 2017-07-29:
Here some additional informations.

I am using a C# UWP app with following references:

Windows Desktop Extensions for the UWP 15063
Windows IoT Extensions for the UWP 15063

and the NuGet library:

Microsoft.IoT.Lightning latest stable 1.1.0

I use the new MinnowBoard Turbot Quad Core with:
Windows IoT Core 15063.0

Thanks and best regards
Sascha

SaschaIoT, 2017-08-07:
The ValueChanged event working with Raspberry Pi 3 Model B and Windows IoT Core 15063 without any problems. So I guess it is a bug in the library in combination with the MinnowBoard.

matt1va, 2018-02-13:
This should be fixed with PR #52

msalehmsft, 2018-02-13:
Thanks @matt1va for submitting the fix. Closing now.

closed 2018-02-13 by msalehmsft