ms-iot/lightning #48: The library does not work with console APP

xplab-cap, 2017-08-21:
The early version of the lib/dll worked with console app.
The actual version can't be loaded, I think because Desktop UWP SDK Extensions are required .
Please remove this constraint, it make no sense to have this limitation for a lib that have to gives access to hardware resources.

Thank You.
Regards,
CAP

msalehmsft, 2018-02-13:
Hi @xplab-cap , the

msalehmsft, 2018-02-13:
Hi @xplab-cap , the Desktop extensions was required in earlier versions of the SDK since the CustomDevice API required to access the device was part of that extension SDK. As of v.16299 (RS3), it's no longer required, so we will remove it once the the DLL is upgraded to 16299 as well.

closed 2018-08-21 by saraclay